bdurbrow/grbl-Mega #2: Safety Door Input misbehavior

bdurbrow, 2019-08-10:
From a report by @zvizdic:

> I tried using a safety door pin as my hold/resume since I have one on/off switch i place (see pics)but it’s not doing anything . Also try the feed hold and it works.
> I would rather have existing on/off switch doo hold and resume
> Any chance for some help
> Switch should be connected to control ground and pinD43 if I am right

_________________________________________________________________________________

OK, I think I found a problem with the safety door handling in the main grbl distribution. I've opened an issue over there; and patched my version here... it seems to be working at the moment.

There's something odd going on with updating the UI during a resume operation; I suspect that there's someplace that I need to drop a call to UITask() into that I haven't found yet.

Note that in order to resume operations, grbl expects both that the door switch has returned to normal; and that a cycle start has been issued (either on the serial port or on the cycle start input pin).

bdurbrow, 2019-08-10:
I've adjusted the handling of the hold state... the UI updating seems to be working now.

zvizdic, 2019-08-10:
Ok where can I download latest software so i can test?

bdurbrow, 2019-08-10:
All updates will be made to the main repository:

https://github.com/bdurbrow/grbl-Mega

Just click the green "Clone or Download" button; re-download the zip; and replace your old files with ones from the zip (obviously, if you made any changes - config.h or cpu_map.h files come to mind - you will want to keep the old files somewhere so you can re-apply those changes to the newly downloaded version).

😄

zvizdic, 2019-08-12:
Ok I got a file and tested it stopped but it doesn’t restart after ungrounded.
Do I understand that right?
If the pin is grounded it will stop ungrounded resume on its own.

bdurbrow, 2019-08-12:
Press your cycle start button after returning the door switch to it's normal position.

zvizdic, 2019-08-12:
O OK Thanks