bdurbrow/grbl-Mega #3: Documentation Discussion

bdurbrow, 2019-08-10:
If anyone finds an issue with the documentation for this fork of grbl-mega; or needs something documented that isn't yet documented --- this is the place to talk about it.

😄

bdurbrow, 2019-08-14:
_**WARNING!!!**_

I just spotted an error in the Wiring Overview that I had up on the wiki... the 5v and ground pins were reversed on the keypad connector (RAMPS Aux 2).

I've corrected the Wiring Overview image and uploaded it - but if anyone has downloaded the WiringOverview.png file; you should delete the old one and download a fresh copy.

Fortunately I don't think it's caused any issues, because the 4x4 matrix keypad is a completely passive device; it doesn't hook up to the power lines at all... but... better safe than sorry!

:flushed:

zickdeko, 2019-09-06:
Wiring Hookup for CNC with RAMPS 1.4
sorry for the google translator text
there is an error in the aux 4 photos with the pins disconnected loses the card,

bdurbrow, 2019-09-06:
I'm not sure I understand...

You are saying cycle start needs USE_UI_ENCODER_B and USE_UI_ENCODER_B_ISR turned on to work?

zickdeko, 2019-09-06:
bdurbrow great job thanks

basic setup
ramps 1.4
lcd
4x4 keyboard

bent aux 4 connectors
sd card not work

cicle start not work

i can't start gcode
interface read from card and can't give circle start.
only change in arduino was to uncomment the lines for the second encoder

bdurbrow great job thanks

coilheaddesigns, 2019-09-06:
Have you gotten the hard limits to work

zickdeko, 2019-09-06:
[image]
sd card not work in this mode

original ramps works sd card
gcode only does not start

double translation is terrible

zickdeko, 2019-09-06:
[image]
[image]

zickdeko, 2019-09-06:
I appreciate the help.
gcode error
first line% symbol
in ugs it works

bdurbrow, 2019-09-07:
Grbl's g code interpreter does not support the % marker. UGS strips that out when sending the file to Grbl.

To make it work from the SD card; remove the %.

I will add a filter to the SD support library to automatically strip it from the file as it is being read by Grbl; but this won't be available to download for a little while. Perhaps next week.

littledrumer, 2019-12-02:
How do the Part Zero, Probe-, Probe+, and LPAB keypad buttons work?

I can not seem to find any documentation for these buttons.

For the LPAB button, I figured out how to get the L and P characters but not the A and B characters. I have not needed these yet but curious how this works. I feel like it is a configuration that needs updating because I see in the UI Support folder you have 3 images and that button is the only one that changes.

Also, I finally found documentation on how to use the jog command, specifically how to switch jog increments. This would be useful info in the wiki as well as some more info in the keypad.

https://github.com/bdurbrow/grbl-Mega/issues/22#issuecomment-546717817

bdurbrow, 2019-12-03:
I'm working on the documentation as I get time; and hope to have some new pages up in the wiki this week.

The A and B functions of that key are only enabled on machines that have a 4th or 5th axis --- but support for that is currently not present in the core Grbl-Mega firmware on which this is based, so you can ignore them for right now.

The UI support layer was written with the intent to support future versions that have 4th and 5th axis functionality; as well as being portable to ARM processors. There is currently a version of Grbl-Mega that does have that support; and at some point I do intend to extend this version with that functionality.

FWIW, this is the 4/5 axis version I'm talking about:
https://github.com/fra589/grbl-Mega-5X

littledrumer, 2019-12-03:
More documentation would be great but no rush, I know its a lot to document.

Adding support for 4tha and 5th axis would be great. Looking forward to seeing that added in the future. The project you tagged looks promising.