bdurbrow/grbl-Mega #7: transfer Gcode via USB

qulet, 2019-09-17:
Bdurbrow
Is it possible to transfer Gcode via USB from Programs on a PC that supports GRBL?
For me, all GRBLPanel functions work ...... not only the G-code transmission over the cable does not work: (((((

bdurbrow, 2019-09-19:
That’s odd... I didn’t think I messed with that. It was working from the Arduino serial monitor... I will investigate.

qulet, 2019-09-19:
bdurbrow!
THANK YOU FOR YOUR TIME

qulet, 2019-09-19:
Excuse me ! I had some errors in the NC code. Today, I once again connected the driver via USB cable. I was sending the code from the programs to the Machine:
GRBLPanel - connects without problems works OK!
Lightburn - connects without problems works OK!
LaserGRB - harder to connect to USB, but eventually connects and transmits OK!

closed 2019-09-19 by qulet