bdurbrow/grbl-Mega #10: Enhancement Request - Switch to use D8 for spindle pwm/laser pwm

thawkins, 2019-09-23:
Would it be feasible to add a compile time switch to move the spindle enable (pwm) to D8 so that for small setups the FET on the RAMPS can drive a laser/motor directly, without needing any interface hardware.

bdurbrow, 2019-09-24:
> Would it be feasible to add a compile time switch

Er... not exactly. As @Saur0o0n mentioned, it's in cpu_map.h; look at line 374 thru 396.

Other machine-specific pin mapping configurations would also be made in this file, should you need them.

thawkins, 2019-09-24:
I will look at and see if i can come up with an option to move it, it just makes making small laser and engraving rigs easier, the FET on the ramps heat bed connection (D8) is quite capabile of pwm driving a 775 or 550 spindle motor or a 2-10W laser.

bdurbrow, 2019-09-24:
Eh... I guess that would be OK. Just #ifdef the requisite lines with SPINDLE_USE_D6 and SPINDLE_USE_D8.

On the one hand, keeping the config files clean is a factor; but on the other hand ease-of-use for the "out-of-the-box" experience is also a factor... and those are popular machines... so... um... yeah.

bdurbrow, 2019-09-24:
Oh, and like I mentioned on the other pull request... I prefer to use FileMerge; so just push your changes to your fork, and I'll take it from there.

Saur0o0n, 2020-01-22:
I'm going to try add laser to my new CNC machine today (so far it was used on vanilla grbl with direct spindle connection - no spindle controller, just MOSFET to change 5V PWM to 12V PWM).
So it was either spindle or laser connected.
 
Now I want to provide additional cable with gnd,12V,PIN 6 signal to the laser controller (it build in laser housing) - along with cables for spindle control/hall feedback. But there will be also spindle controller connected all the time directly to spindle PWM (pin 6),enable,direction pins and power lines inside the CNC controller box.

Assuming I'll connect laser only when used, this will still make spindle controller working and trying to power not connected spindle, when laser is in operation. I'm not sure if this is any kind of the real problem, but it definitely does not feel right. Spindle controller will be in error state all this time.

As a remedy, I think we could either move laser to different PWM pin, or simply perhaps, do not enable spindle while laser is in operation - the second approach looks much simpler and easier. My two lasers does use spindle_enable signal, I'm not sure if it's true to all other lasers market - but probable yes.

What do you think?

ps. I've checked and when is $32=1 spindle_enable pin goes high on M3/M4

Saur0o0n, 2020-01-24:
I've created pull request with small code change that introduces config.h switch DONT_SWITCH_SPINDLE_ENABLE_FOR_LASER.
If enabled, it prevents changing SPINDLE_ENABLE state while $32=1.
Since it's compilation time directive, while disabled it does not alter the original code - so should be perfectly safe.
Tested on my machine.

bdurbrow, 2020-03-05:
I'll look at it when I get a chance.