bdurbrow/grbl-Mega #11: Information - SDCard initialization can correctly drive FlashAir Wifi SD Card

thawkins, 2019-09-23:
This is not really an issue, its just a piece of information.

Unlike marlin, grbl-Mega seems to correctly enable the SD Crad in such a way that the Toshiba FlashAir Wifi SD card works correctly and can be connected to: This works for most 3d Printers but for some reason has never worked correctly with Marlin. 

See the article i wrote on the subject of setting it up for a 3d printer on:

http://microfabricator.com/blogs/view/id/54e438e56983c5316480205e/using-a-flashair-wifi-enabled-sdcard-with-your-3d-printer

bdurbrow, 2019-09-24:
> This is not really an issue, its just a piece of information.

All _Bugs_ are _Issues_, but not all _Issues_ are _Bugs_... 😏 

In the credit-where-it's-due department... I did not write the SD library that I used myself; I just ported it to run in the Grbl-mega environment. Parts of it came from the Arduino project; parts of it were contributed by SparkFun. I did not _intend_ for this to work... but since it does... BONUS! 😄 

From your article:

> By contrast sending that amount of data across a USB 2.0 link to an Arduino or other 3D printer controller can take up to 30 minutes.

That's because most 3D printer controllers are based on AVRs that lack a native USB interface... there's some USB to serial port chip sitting in the middle, and those are comparatively slow... even cranked up to the max baud rate that they support. The ARM port of Grbl that Chamnit is working on should help with this issue; should one wish to have their CNC tethered to a PC/Mac (I don't - which is why I have it reading from an SD card).

Hmm... Amazon also lists a bunch of other SD/WIFI cards, some for about half of the price of a Toshiba.... do you know if any of those work also?

thawkins, 2019-09-26:
I have not had any experience with anything other than a FlashAir, i managed to het two of the version 2 cards for next to nothing,  i have heard stories about other cards that indicate they may have reliability issues, the FlashAir is rock solid.  I have one in my 3d Printer and now one in my CNC rig, i live in a condo and space is at a premium, so my machines live on a workbench on my balcony, so getting connections to them is tricky, the wifi cards make that simple, and allow me to send files from any .

Im thinking of adding a Raspberry PI instead of the FlashAIR running OctoPI with the gerb CNC plugin, that way i can remote control the machine via Wifi, and have it stream video from a cheap camera back to me. 

I probaly would not leave the CNC rig unattended,  the 3D printers are ok with that, but therenis too much that can go wrong with CNC.

bdurbrow, 2019-09-26:
> so my machines live on a workbench on my balcony,

What's the climate like where you are? Here, I'd be worried about them getting ruined during a winter storm...

> Im thinking of adding a Raspberry PI instead of the FlashAIR

_Should_ work... I deliberately left the serial IO system intact... If you find any problems, please let me know about it so I can fix them.

Saur0o0n, 2019-09-26:
There are combo boards ARDUINO MEGA2560 R3 ATmega328 + WiFi ESP8266 on single PCB - pin layout in Mega compatible. This could add simple way to use WiFI with grbl.
I currently use ESP32 for my [PID controller](https://github.com/Saur0o0n/pidkiln) for kilns - with full HTTP server - and it works quite well (it's much more powerful then Atmega in any way - just doesn't have as many pins as 2560 :) )

[image]

bdurbrow, 2019-09-26:
I should probably order one for testing... I should be able to set it up so that you can send a file over the ESP8266 - perhaps via wget or curl, as the last step in a postprocessor script - and cache the file on the local SD card (this would eliminate any timing 'hiccups' in reading the file over the WiFi link). Hmm....

Saur0o0n, 2019-09-26:
ESP has plenty of RAM and cpu power (2 cores) + local memory/disk (up to 4MiB) called SPIFFS. So you/we can easily use it to prepare webserver with webpage to all uploading/setup changes etc. Store in on local SPIFFS and then in background feed it to sdcard - or perhaps even dirrectly from SPIFFS to Atmega for process (I don't know how exactly this board is constructed).
I'm up2date with webserver and/on esp - so I can help/write this part.

bdurbrow, 2019-09-26:
Hmm... I don't think we should be storing g-code files on the SPIFFS. It's only 32MB; and I wouldn't want to wear it out (I'm not sure how good the flash is that ships on those cards, but I suspect it will die long before a good SD card will). Also, a SD card is easy to replace - in practice, for most users, fixing the soldered-down flash is going to mean replacing the whole board; and re-doing the configuration for the machine.

Saur0o0n, 2019-09-26:
Well it's something to consider - but I've been also worrying about wear out of this flash and Espressif claims it's using chips that will withstand at least 100 000 writes. So this is a lot, for this type of usage - don't believe singe user can wear it out.
Probably the same chips are used for PSRAM (pseudo ram) - this can be also next 4-16GiB to malloc data (depending on used board) - but I guess this Mega+ESP8266 combo won't be so generous in flash.
Also SPIFF is directly accessible from ESP - SDCard isn't - at least with current MEGA configuration. So this makes it good intermediate storage, and in my opinion, better storage because I do not like to swap sdcards in my printers anyway (this is my experience - I've been driving my CNC router directly from computer). So uploading with http seems to me much more flexible. There is nothing against using both storages ... as raid :D - kidding, just two storages :)