bdurbrow/grbl-Mega #9: Simple docs, howtos

thawkins, 2019-09-23:
i have my system up and running now, with a reprap discount card, but i cant work out to do some basic things. 

I don't have a keypad connected, and would prefer not to do so. I would prefer to use the encoder on the RepRapDiscount 2004 to select menu options. I have tried pushing the encoder push button but that does not do anything. 

I have tried putting an SD card in with a test.nc file as the only file, its a 4GB card, 

tasks i would like to work out how to do are:

1. List files on SD card and select one for running
2. Home the Machine
3. Return to zero
4.Toggle M7, M8 (enable coolant), (enable Misting)
5.Toggle Laser Mode
6. Enable laser on low power setting to allow focusing.

bdurbrow, 2019-09-24:
> I don't have a keypad connected

That's the problem. A 4x4 matrix keypad is required equipment.

> and would prefer not to do so.

Why not? Space on the front panel?

I'm afraid you are going to have to add a keypad - the entire UI system was/is designed around having one. However, it doesn't have necessarily have to be a large keypad - there are small keypads built with tactile switches available from the usual sources (eBay, Amazon, Aliexpress, BangGood, etc). Just be sure it's a 4x4 keypad with an 8-pin matrix output, and not a 4x3 keypad, or a serial one.

In the "UI Support" folder of the source code download, there's a PDF showing the key layout. It's sized for those adhesive-backed 4x4 membrane keypads from China; but can of course be resized to fit something else.

Also - the Shift key acts more like the Caps Lock key on a PC keyboard does - push it once to activate it, and again to cancel it. Pushing any other key on the keypad also deselects Shift. Due to the way matrix keypads are built, holding down the shift key and pressing another key will cause the system to mis-read the keypad (the "ghost key" effect) - this is why it works the way it does.

>    List files on SD card and select one for running
Main menu -> "SD Card"
In other words, press the Menu key on the keypad; then select "SD Card" from the main menu.

>    Home the Machine
Main menu -> "System" -> "Home All Axis"
Note that the homing cycle must be set up and working for this to work - all the menu command does is issue the same homing command that a serial-port based controller (like UGS) would.

The main Grbl wiki has info about doing the setup. Also, be sure to look in the wiki here for the pinouts for RAMPS - what's marked on the PCB is NOT where you should hook everything up to.

https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle

>    Return to zero
I don't have a specific command for that yet... so for now; either use the jogging controls or enter on the MDI command line G0X0Y0. You will obviously want to make sure that your Z axis is clear of anything that might be in the way...

>    Toggle M7, M8 (enable coolant), (enable Misting)
MDI again. Alternatively, you may wish to rig up some rotary or toggle switches to select between having the coolant solenoids or pumps controlled from Grbl, and having them be manually selected off or on (in other words, put a switch between the output from Grbl and whatever transistor card or relay that turns on the power to the coolant system).

>    Toggle Laser Mode

DISCLAIMER: Please know what you are doing with a laser! A proper enclosure, interlock switches, and safety gear is a must. Blindness, injury, fires, and other bad things can happen in an instant if the beam gets loose. Also, do remember that this is PRE-ALPHA software, and might misbehave!

That said...

I haven't had a chance yet to do anything with lasers. I have a K40, but it's in pieces right now... it probably needs a new tube due to age, and in any case I need to integrate a Grbl-based controller and put it back together.

Anyway... laser mode is - at the moment - a configuration option that must be set over the serial port.

More about it:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

>    Enable laser on low power setting to allow focusing.
I think you can just turn on the "spindle" while in laser mode at a low "RPM" to get what you are looking for... so, use the MDI command line?

thawkins, 2019-09-24:
No front pannel....

thawkins, 2019-09-24:
What if i just wire a single button across the pins for "main menu", will
that allow me to do it?,  cabt you use the built in switch on the encoder
to activate the main menu.

bdurbrow, 2019-09-24:
> No front pannel....

Um... I'm not quite following? You have a 2004 LCD card... mounted how?

>What if i just wire a single button across the pins for "main menu", will
that allow me to do it?,

Well... yes, but you're missing out on a bunch of stuff... if you at all possibly can you really should have a keypad.

> you use the built in switch on the encoder to activate the main menu.

Well, OK, I've added that behavior to the StatusUIPage; it'll be in the next update to the repository, but... without a keypad you:

- Can't jog or power-feed.
- Can't use MDI.
- Can't use probing.
- Can't use the part zeroing functions.
- Can't use the RPN calculator (depending on what kind of machine you have, this may or may not matter to you).

thawkins, 2019-09-26:
I have decided to take your advice and design a side panel that holds the keypad,  cycle control buttons, and 3 pots for the overrides. I dont want an mpg, as its too big, so im thinking of using a standard rotary encoder, although it only has 3 pins, i belive its the same A/B quadrature output just with the A- and B- pins joined together. 

I will design the pannel to replace the lefthand side brace, so it will also act as a brace. I have ordered some nice 4x4 key matrixes that are made from physical switches. 

See: 
https://s.lazada.com.ph/s.Am63

bdurbrow, 2019-09-26:
> i belive its the same A/B quadrature output just with the A- and B- pins joined together.

Assuming that we are talking about the same switch-type encoder knob that's on the LCD card -- it's not the same as a MPG wheel. It's quadrature; but the third pin goes to ground; with the pull-ups on the input lines of the AVR turned on.

I have not _tested_ that configuration; but it _should_ work, as it's the same code that's used to read the encoder knob on the LCD card. If it doesn't work, please let me know so I can fix it!

However - if you leave the second encoder turned off in the config files; it will use the one on the LCD card for jogging, so you don't _have_ to have it. Only the keypad & LCD board is mandatory.

> I have ordered some nice 4x4 key matrixes that are made from physical switches.

That's the small one I was thinking of. I ordered a few when I ordered the adhesive backed ones shown on the wiki. Unfortunatally, they came fully assembled - with that 8-pin header sticking up in rather inconvenient position. Probably obvious, but I would suggest de-soldering the header and flipping it over to the back side of the board; if yours comes the same way.

thawkins, 2019-09-28:
Yep thats what i was planning, I made a trip out to our local electronics parts store today, and got an Mpg, 3 loverly 10 turn 5k linear pots, and 4 cool durable panel switches. 

I also got 2 of the flexable keypads you show. 

I redesigned the right hand brace on the cnc to have a broad panel on it with all the right cutouts, it sticks 60 mm out the side of the machine, but it ok, because that side has the X stepper on it which also sticks out by 60mm.i also got some nice 10 turn vernier knobs for pots with a turn counter and a lockoff lever. I will post some pics when i'm done.

bdurbrow, 2019-09-28:
Hmm... ten turns.

Well, I suppose it's a personal ergonomic decision, but you _might_ discover that you want the faster responsiveness of a standard 270-degree pot.

There's a couple of those on my old Tektronics AN-425 oscilloscope... and it while it's great for (literally) dialing in a precise value; it sure does take a lot of turns to do so...

In any case; I'd love to see what you come up with; and how the ten-turn pots work out for you in practice.

😄

thawkins, 2019-10-05:
I switched to 270 degeee pots,  mainly becausee tbe vernier knobs for the 10turns where hideously expensive. My ribbon cable and mpg turned up today so im going to be getting it wired up, and then continue work on the 5ach sensor. 

[image]

thawkins, 2019-10-05:
My local kinkos/printshop did a wonderful job of printing a sheet of keyboard stickers of the keyboard, scaled to exatly thethe right

thawkins, 2019-10-06:
Ok, finaly got my machine fully wired and tested, been cuting aluminium plates from sd card all day, i added two more system menu items "return to zero" and "set zero pos". As i needed them for my workflow, and with them i can operate the cnc rig completly without an attached computer. 

I only have the probe and the tach to sort out. The probe is wired, just not tested.  I will switch back to workingnon a reliable tachometer probe now. 

There a are a couple of small ui issues,  lack of a back button/function to back out of a chosen action, like viewing tne sdcard, and deciding tbe file you want to cut is not on there. 

The override pots are great, the finaly gave me the courage to ramp up my feed rates on cutting aluminium. 

One thing i did nitice is i tried to use tne MDI to executr a "M7" to turn on my blower, but nothing happened, sending m7 via the usb worked fine.

bdurbrow, 2019-10-06:
> There a are a couple of small ui issues, lack of a back button/function to back out of a chosen action, like viewing tne sdcard, and deciding tbe file you want to cut is not on there.

Push the button under the knob on the LCD PCB.

Yes, I need to write up a wiki page documenting how to navigate the UI and find everything...

> One thing i did nitice is i tried to use tne MDI to executr a "M7" to turn on my blower, but nothing happened, sending m7 via the usb worked fine.

That's odd... all the MDI page does is send the entered command line directly to the main g-code interpreter; the same as the serial port does... and I just checked it on my machine; it's working here. Did the "M7" line disappear from the LCD? If it didn't; then that means that the "Enter" key-press didn't get read by the Arduino...