bdurbrow/grbl-Mega #14: Enhancement, Laser mode pin

thawkins, 2019-09-28:
Would it be possible to have a pin that goes high when lasr mode is enabled. This would allow me to switch laser power though a relay so it is only enabled in laser mode.

bdurbrow, 2019-09-28:
In principal, yes... any particular pin you'd like as the default in the cpu_map.h file?

Also, be aware that in the current setup; IIRC laser mode is saved to the EEPROM; causing a write cycle every time it's turned on or off.

When I implement this; I'll see about a compile-time configuration flag to disable the write for that parameter.

thawkins, 2019-09-29:
Im not fussy, anything free will do.....

thawkins, 2019-09-29:
Having thought about it,  there is a spare pin on the block with spindle pwm/dir/enable 

That is probaly the best place

We also should ensure that spindle enable cannot not go high at all when in laser mode.

thawkins, 2019-09-29:
re the eeprom write, that should probably be changed to only write the eeprom on a $x like everything else.

bdurbrow, 2019-09-30:
>  there is a spare pin on the block with spindle pwm/dir/enable

That pin is going to be needed for a spindle tach pulse input.

Next to it, on the AUX 1 port, there is another unused pin - D58. So, that's what I'm going to set as the default for the Laser Mode Active pin.

Of course, one can re-arrange things to one's requirements by editing the cpu_map.h file.

> We also should ensure that spindle enable cannot not go high at all when in laser mode.

Um... yeah... that won't work. Some machine setups use that output to turn the laser on.
For dual mode devices; I think that handling this in hardware is probably the better choice - either with a relay or an AND gate triggered off of the 'Laser Mode Active' pin. However; I think I'd be less worried about activating the spindle inadvertently - which probably wouldn't have a cutting bit in it, if one were using the machine in laser mode - than inadvertently activating the laser; which can be a nasty hazard if you aren't expecting it.

This is another good reason to have a laser machine in a full enclosure with safety interlock switches.

> re the eeprom write, that should probably be changed to only write the eeprom on a $x like everything else.

That's the problem: you enable laser mode with a $32=1 command... which then promptly writes the whole settings struct to the EEPROM... each and every time _anything_ (including laser mode) is changed.

So... what I've done is to remove the unconditional call to write_global_settings() at the end of settings_store_global_setting() and put it in `case 255:` of that function - sending `$255=1` will cause the current settings state to be written to the EEPROM.

Saur0o0n, 2019-09-30:
> So... what I've done is to remove the unconditional call to write_global_settings() at the end of settings_store_global_setting() and put it in `case 255:` of that function - sending `$255=1` will cause the current settings state to be written to the EEPROM.

Just document it clearly - since it can bring a lot of grey hair for some, trying to find out why machine settings does not save :)

bdurbrow, 2019-09-30:
> Just document it clearly

It'll be in the Readme.md file that shows up on the main page of this repository. It's the first item listed for the October update.

bdurbrow, 2019-11-07:
Closing this issue now because the requested functionality was included in the October 2019 update.

closed 2019-11-07 by bdurbrow