bdurbrow/grbl-Mega #15: Enhancement:Variable spindle speed

thawkins, 2019-10-05:
I have been reading up about speeds and feeds and chatter, and found an obscure reffernce to a simple mechanism for reducing chatter. Chatter is caused by resonances building up between the tool and the workpiece, which degrades the cut surface, that in turn can result in an uneaven cutting surface that further enforces resonance, its a posative feedback loop that can severly degrade surface finish, and in extreem circumstances start to cause tool deflection that can result in a broken tool. 

If you have a mechamism to vary spindle speed by say +/- 50 rpm over 2-4 seconds, it can stop resonances starting and kickstarting the whole process. 

I have an 8khz timer running on the tach, that could be used to perform that variation, the Hass machines use $165 for the varience value, and $166 for the timespan.  They call it SSV.

Saur0o0n, 2019-10-05:
This could be implemented directly to GRBL - in spindle (and laser too) mode RPM is regulated with PWM. When I did tests with laser it was 1-2Khz (it's variable with timer). So from what you are saying, it would be enough to provide some fluctuation in time here.

bdurbrow, 2019-10-05:
Not hard to do; calculate the desired variance; and then call `spindle_set_speed(spindle_compute_pwm_value(gc_state.spindle_speed + variance));`

I think instead of doing it as a $ setting; I'll implement it as an M code... as you may want to turn it on or off for each individual program, tool, or even region of the part being machined.

Another thing that could be done is to attach an accelerometer - or even just a microphone - to the frame of the machine; feed that into a separate processor to do some DSP work; and have the main processor avoid RPMs that generate resonance.

bdurbrow, 2019-10-05:
Also... look in DriverUtilities.c; Timer 2 is setup as a 976hz system clock. (yes, I just spotted that the comment says Timer 5... but the registers being setup are for Timer 2... the comment fix will be in the October update).

I think that due to the low update rate needed to vary the spindle RPM; calling the routine from the same locations that UITask() is called probably makes more sense than hanging it off of an ISR.

bdurbrow, 2019-10-05:
> I have an 8khz timer running on the tach

Yeah... the way I've got my preliminary tach implementation setup only calls the ISR for pin change events (which as mentioned in the other thread only happens on the order of a couple of hundred hz - 6000 RPM being 100 revs per second); and an overflow ISR once every 4.2 seconds if the spindle is turning so slowly that the Timer 5 counter actually overflows (the pin change ISR resets the counter to zero).