bdurbrow/grbl-Mega #16: Enhancement: pwm on m7 and m8

thawkins, 2019-10-05:
This is not an urgent requirments, its more a placeholder request. 

Currentl m7 and m8 enable and disable the coolant and flood funtions, i have 12v centeralfugal blower on m6, and i have aquired a 12v liquid  pump and misting setup for m8, both are capable of variing thier response based on pwm.  At the moment they can only be turned on and off (D9 and D10).  Having the ability to set the output via pwm would be cool,  but would require a modification to the gcode. Alternatly if enabled the systen could use the current Sxxxx value so 

S500 M7

Would set the coolant pump to 50% pwm. 

We could hijack a $nnn value to set the m7, m8  behaviour.

bdurbrow, 2019-10-05:
S should always control the spindle; however there are other designations that could be used.

Both of those pins are on Timer 2; which is being used for the 976hz system clock. It shouldn't be much work to turn PWM on for those.

erosenst, 2020-12-14:
You can use laser mode to make pwm. $32=1, $30=1000, then m4, S500 and when there is movement "hotbed" output pwm 50%.