bdurbrow/grbl-Mega #17: Enhancement, show effective feedrate and spindle

thawkins, 2019-10-06:
2nd page of the status monitor shows the F (feedrate) and S (spindle speed) values, but these dont show the acfual values after overrides, perhaps put the actual value with a dash to seperate it so tbe effective rate can be seen.

bdurbrow, 2019-10-06:
There isn't enough room on page 2; but there is on page 3 if I adjust how the values are displayed. Note that I'd like to keep the display format compatible with processors that have higher step rates (and thusly higher maximum expected feed rates) such as the ARM version that @chamnit is working on.

bdurbrow, 2019-10-06:
Now that I think about it; it would be nice to have the actual axis speed along with the actual spindle speed displayed together on a single page of the status display. If the laser-mode-active output is disabled; that pin could also be used to read an analog current sensor to display the spindle load on that status page as well.

bdurbrow, 2019-11-07:
Closing this issue now because the requested functionality was included in the October 2019 update.

closed 2019-11-07 by bdurbrow