bdurbrow/grbl-Mega #20: New Machine Setup Questions

bdurbrow, 2019-10-26:
From a comment by @Elpablito:

> Hi bdurbrow
> At first congratulation for the fantastic job you have done, I am following you on github since last year when I start to think to make CNC.

Stop that! You are going to over-inflate my ego! :smirk:

> Sorry for my English I shall try to make my best. I made a machine, bat it is my first one, so I am not an expert in working with such a kind of machine. The CNC is made with some plastic parts good for wood or soft materials. Actually, I have used an Arduino Uno with its shield for stepper drivers and it seems it is working properly, at least for the tests I have done. For sure, since the beginning the idea was to have a standalone CNC so I am trying to replace Uno with a Mega that I was having somewhere. Unhappily, I faced these kind of problems that I am not able to solve.
> 
> 1. I have modified the hardware setting $0……$132 with UGS and when saved. If I disconnect the board and then reconnect the setting are the old ones not the new ones. It seems that save action do not write EEprom.

I'm going to assume that you are using the October 2019 update version.
In the main readme.md file that shows up on the main page of the github repository; where it talks about what's new in the October update; on the fourth item down; it covers this. Basically, send a $255=1 command via the serial port to save your settings; or use the settings editor found in the System menu. The reason this was changed was to accommodate a user who was going to be switching back and forth between regular and laser modes. With this you can switch any setting - including $32, laser mode - and have it not do a write cycle on the EEPROM (and thus wearing it out prematurely) unless you actually want to save the current settings.

> 2. Why in the home cycle homing Z is optional and why is the third action, moving before Y and X and then Z? If any reason I shall prefer to move Z as first action.

I haven't yet done anything significant with the homing cycles; it's more-or-less as I found it in the official Grbl-mega version. It's something that does need addressing; and I think will get to it as part of the next update.

You're right though... that is strange that it's setup that way.

> 3. Which is the use of RAPID_OVERDRIVE potentiometer?
> 4. The FEED_OVERDRIVE Potentiometer is to decrease the set speed of FEED?

The override potentiometers allow you to make adjustments to the feed rate, spindle RPM, and rapid speed (how fast the machine moves when you give it a G0 command). These adjustments take effect immediately, while you are running a program (so if the machine starts chattering or squealing you can make adjustments to get rid of the problem). They are optional equipment, but I do recommend having them. You enable that feature in the `config.h` file, down towards the bottom of it.

If you don't have them setup, you can still adjust the overrides via the serial port.

> 5. Is the SAFETY_DOOR the door of the cabinet you need to close before run or it is an escape way? In such a case what is it doing?
> Thanks

Basically, yes - it's the switch that's on the door of CNC machines that are in an enclosure, and it stops GRBL from moving the machine while the door is open. Or at least it's supposed to -- depending on a variety of factors, it can still fail, so people should always be careful and not put themselves in a position where the machine can hurt them.

Elpablito, 2019-10-26:
Hi
Thank for the answer. It was clear I lost same pieces on the way I had to read again your README to find the lost pieces. Give me some time I have to make a box and put all I have done on same paper, not trusting any more on my memory. If it will be of some interest, I shall send a copy of the job done.  I will use Ramps1.6 because I have it.

bdurbrow, 2019-10-27:
Just a reminder; since a few other people have been caught by it:

The LCD _**MUST**_ be a character LCD; 20 wide by 4 tall. The search terms I used on eBay were "2004 ramps lcd".

Graphic LCDs will NOT work.

Elpablito, 2019-11-07:
Hi                                                                                                                                                                                                 As you can see, I have finished the box and now I have to make the connections. A question raised with Mega and Ramps that is how to connect the two steppers I have on Y axis. I am working with steppers at 24Volt 1Amp and DRV8825. I prefer to have two drivers with one motor instead of one driver with two motors, 2Amp is near the limit of the driver and increasing the voltage is not convenient. I do not know, if I lost again a step in the configuration or if I have to parallel the two steppers. 
[image]

bdurbrow, 2019-11-07:
> I prefer to have two drivers with one motor instead of one driver with two motors

I'm suspecting that what you wrote isn't _exactly_ what you meant. What you should do is one driver per motor.

For the Y axis, you have two motors. On the RAMPS board, you have 5 driver sockets; X, Y, Z, E0, and E1.

So, put a driver into each of the X, Y, Z, and E1 sockets. Hook the extra Y axis motor to the E1 driver.

Then, in cpu_map.h, on line 183, enable the CLONE_Y_AXIS option.

Leave the SQUARE_CLONED_Y_AXIS option turned off for now; it's totally untested and probably doesn't work right.

The default setup when you turn on the CLONE_Y_AXIS option is to map the cloned Y axis to socket E1 on the RAMPS board; likewise, turning on CLONE_X_AXIS would map the X axis onto the E0 socket.

P.s. You are probably aware of it, but just in case: to run the RAMPS board at 24 volts, you need to modify it slightly. The RepRap wiki has more info:

https://reprap.org/wiki/RAMPS_24v

Hope that helps! 😄

Elpablito, 2019-11-08:
Hi
As I said my CNC isat the moment working with arduino uno and CNC shield. Picking up a DRV8255 from UNO e installed on ramp I have no Vref, the same driver back to UNO is working properly.  I have done the test supplying Mega with ramps installed without any other connections. The voltage on the pin of driver under test are the same of the drivers installed on  UNO and they seems to be ok, but Vref = 0. It is like the internal 3.3 volt does not exist. Have you any suggestion? Thank again

bdurbrow, 2019-11-09:
What's the enable pin read on your multimeter? I'm just guessing, but perhaps the driver module doesn't supply voltage to that Vref potentiometer unless it's enabled???

Elpablito, 2019-11-09:
Hi
Your suggestion was my first thinking because the voltage on pin enable is 5.08V so I measured a driver on Uno but the result is the same 5V. In detail these are the voltage on driver
Enable = 5,08V                                                                                                                                                                                                                        M0 = 0V                                                                                                                                                                                                                                                                                                M1 = 5V    there is a jumper
M2 = 0V
RST = 4,54V   Connected just with SLP  is for sure  the pull-up of  the DRV8825
SLP =  4,54V
Step
Dir = 0V
GND     tested it is really to ground    
Fault = 5V
----------
Vmot is not connected
 
With some small differences are the same on  Uno.
Before to ask for a help I was looking for an answer for same time then I said “OK magic I will ask”

bdurbrow, 2019-11-10:
Hmm... I'm just guessing, but --- on most DRV8825 modules that I've seen; enable needs to be low to activate the driver; and you've got it showing up as high.

Just for setup; try setting $4=7. This will invert the enable pins going to the drivers, so that they are enabled while the machine is sitting idle.

After you have the current set, you will want to set $4 back to 0.

bdurbrow, 2019-11-10:
> Vmot is not connected

Actually, I think that may be your problem. The DRV8825 driver gets all it's power from Vmot... including the power to the current set potentiometer.

Elpablito, 2019-11-12:
Hi
I send you a photo of the CNC I have done so you can understand me. This is my first job and I have a lot of thing that I do not know.
 At first the problem of Vref is been solved. It is necessary to connect the motor voltage to be able to read the Vref on ramp.  (This is not necessary on CNC shield and UNO)
I have not a proper spindle, but a DC motor with a tachometer wheel controlled by a separate device.  

Display and keyboard are working.

I followed your formatting instruction, and now I have to find a test file just to check if SD card is working?

I have uncommented the following line
 #define UI_FEED_OVERRIDE_POTENTIOMETER	       	13  	//connected to A13 5K POT on ramp T0	
 #define UI_RAPID_OVERRIDE_POTENTIOMETER	14	//connected to A14 5K POT on ramp T1
 #define UI_RAPID_OVERRIDE_NOMINAL_MAX_RATE     	500.0   
Is it possible to check if they are working without connecting steppers?

Connection of push button
START  		D32       AUX4 pin 3	GND AUX4 pin2
RESET  		D47	AUX4 pin 4	GND AUX4 pin2
FEED HOLD	 D45	AUX4 pin 5	GND AUX4 pin2
I do not know if the connection are correct, but pushing the reset, I do not see any change on display
 
How can I check if hand wheel is working?    
Thanking again for your help                                                                                                                                                                 
[image]

Elpablito, 2019-11-13:
Hi
I have connected some spare steppers so I solved all the question of my previous post It is just a matter of knowing  which button you have to push. The only thing that I have not tested is SD card, but before I have to learn how to make a Gcode file.

Elpablito, 2019-11-18:
Hi
I wrote a .g-code file on an SD card formatted as suggested by you, but I am not able to read it. The system see an SD card, bat not the file in the card. Which kind of file can you read?

bdurbrow, 2019-11-18:
The SD driver can only read filenames with up to 8 characters in the name; and the firmware expects an extension of `.nc`

This should work:
`test.nc`

But this won't work:
`This is a test program file.nc`
or
`test.txt`

Elpablito, 2019-11-19:
Hi
I include a file test.nc made with freecad because I never saw such a kind of file so I am not sure if it is formally correct. After I select the test.nc pushing the encoder on LCD. I pushed the button START D32 connected to AUX4 pin 3 to see the steppers move, but this does not happen. Have you a suggestion? Thank

[test.zip](https://github.com/bdurbrow/grbl-Mega/files/3866078/test.zip)

bdurbrow, 2019-11-21:
That’s... odd. I’ve looked the file over on my iPhone; it looks functional (it’s got some strange extra moves in there and it keeps switching from absolute to relative positioning mode, but it should _work_).

I’m just guessing but there might be a “bad” character in the file that Grbl is hanging on... if so I’ll add it to the list of stuff that I filter out when reading off of the SD card.

I’ll check it on an actual Arduino running Grbl as soon as I get a chance.

The .nc file is just a text file with g-codes in it; you can open it in any text editor (Windows Notepad, Notepad++; TextEdit, BBEdit, emacs, vim, nano, pico, whatever you have handy).

Elpablito, 2019-11-21:
Hi
Solved I can go head. The problem is this comment (; G98) it stop GBRL, but not ask me why.

bdurbrow, 2019-11-21:
Sounds like a bug in the character filter... GRBL itself can’t handle anything but pure g-code; it depends on whatever is feeding it to filter out comments and other non-supported stuff. I do this in the file reading code; but I’ve obviously missed something.

Elpablito, 2019-11-21:
Hi
Long time ago I said “ if you want to test something give it to someone who does not know how it works”. 
Tested Slot, Drill single hole and Bolt hole circle. OK very good. Face is not clear; I think it is a routine to make also a pocket. I make an example so you will be able to understand me easily
Start 	X  10
start 	Y 10
size X 	5
size Y 	5
Y step    0.5
Bi_Dir 	[*]
Clear Z	 -20
Start Z	 -30
Depth	 -2
stepZ 	0.5
rpm 	1000
feed	 50
I was thinking to see CNC remake the cycle at Z -30,5  Z -31  Z-31.5  Z-32 and then stop. Instead it make just one cycle at Z-30. 
Is it possible to delete asterisk in Bi_Dir to have the cutting only in one direction? If yes how

bdurbrow, 2019-11-23:
Yeah, I need to write some documentation.

Face is not intended to make a pocket; it is intended to flatten a more-or-less rectangular piece of stock. It doesn't cut any outside walls like a true pocketing operation would.

Depth is a relative value; and I probably ought to be taking the absolute value of it. I'll add that to the next update.... but for now you can add this to `UIConversational.cpp` at line 112:

`values[faceOperationForm_Depth].f = fabs(values[faceOperationForm_Depth].f);`

Hmm... spotted a couple of other minor bugs in there also...

>  Is it possible to delete asterisk in Bi_Dir to have the cutting only in one direction? If yes how

"Checkboxes" like that are toggled on and off by pushing the encoder knob on the LCD card.

littledrumer, 2019-11-28:
I am having trouble running a .nc file from the SD card. The same file works running through UGS but on the SD card, I can not seem to get it to run.

From what I understand once I load the .nc file I want using the LCD screen all I have to do is press the cycle start button and then the program should run. When I try this nothing happens. I have UGS connected to watch the verbose output to check and see if the cycle start button is actually registering as pressed and it is but the file still does not run. I have also tried with UGS disconnected. What am I missing?

I have also noticed strange behavior when inserting an SD card. The LCD goes blank and the only way to get it back is to reset the mega. Is this normal behavior? Should the SD card be hot-swappable?

bdurbrow, 2019-11-28:
@littledrumer -

Select the .nc file, and then press the "Enter" key or push the Encoder Knob in. The file should then start running.

However, as this seems to be an issue causing confusion, I will alter the behavior in the next update to include pressing Cycle Start as another way to get the file running.

As for the SD card causing a crash on swapping - that should only be a problem when a file is actually running and you yank the card. Inserting a card should never do that... which leads me to suspect a hardware problem; perhaps with getting noise on the power supply lines?

Note that monitoring the card presence line to refresh the directory display when cards are swapped is not yet implemented.

littledrumer, 2019-11-28:
@bdurbrow thank you for the feedback.

I am tried running the file as you instructed but it is still not working. I am going to test some more this weekend to see if I can figure it out and I also got another smart controller LCD to see if its a hardware issue. Other than the file not running everything about the smart controller LCD seems to be working while navigating menus and selecting menu options with the encoder button. Maybe it is just the SD card portion that is not working.

I also made sure no file was selected and tried swapping the SD card and it is still crashing so this also points to an SD card being the issue.

Everything else so far is working great! I am trying to upgrade my X-Carve to be stand-alone.

Elpablito, 2019-11-28:
Hi
At first, thank for the fast answer. I have just finished replacing the Arduino UNO CNC control with the new one, so I report my problems hopping they will be of help for someone.  
-	At the switching on the HOME cycle does not cancel the alarm. After HOMING you have to clear the alarm to go on.
-	Limit switch. I select hardware limit switch. They are active only on the HOMING cycle. This is a problem for a not skilled man like me, but I do not know is may be a configuration problem.
-	In Config.h I have modified the lines 108 -110 as follows to have the Z axis going up before move X or Y
 #ifdef DEFAULTS_RAMPS_BOARD
         #define HOMING_CYCLE_0 (1<<Z_AXIS)   // Home X axis (modifica1)
  #define HOMING_CYCLE_1 (1<<X_AXIS)   // Home Y axis (modifica1)
  #define HOMING_CYCLE_2 (1<<Y_AXIS)   // OPTIONAL: Home Z axis (modifica1)
-	SD card.  I made same files with FREECAD and PATH. In PATH there is a postprocessor GRBL but it may happen that same instructions are not compliant with GRBL. In such a case, you can read the name of the file bat it does not run. You have to edit the file and cancel the wrong instructions to have the file running.
-	I am not able to discover how to make a probe cycle. I used command G38.2 without any result.
-	Keyboard. How work “PART ZERO” and “L P”?
-	 Same problem how work CYCLE_START and FEED_HOLD push buttons?

bdurbrow, 2019-11-29:
@littledrumer -

The software doesn't even initialize the SD card until you access the "SD Card" menu item in the main menu... so, yeah, probably a hardware issue.

@Elpablito -

Homing & alarm behavior was inherited from the official Grbl-Mega repository... and, yeah, I think that it could stand some improvement. I'm thinking that being in a non-homed state shouldn't trigger the general alarm, but just lockout movement operations until it is homed. Successful homing should automatically reset that condition; but not clear the alarm state for any other reason. This will mean adding a separate variable to track this.

On RAMPS (and similar boards) the limit switches are only used during the homing cycle, because the interrupt pins are needed for other functions. Instead, set up software limit checking with parameters $130 to $132.

The default homing sequence has been adjusted to make more sense. This will be rolled out in the next update (it was another thing inherited from the official Grbl-Mega).

I'm still trying to figure out what's wrong with the g-code that you generated... it should have just worked; but it isn't.

Probing and Part Zero are a little complex, and I need to write documentation for them.

The "L P" button is used on the MDI screen to type the L or P letters. Press it once, you get "L", press it again and it switches the "L" to a "P".

Feed Hold pauses the machine's motion; and Cycle Start resumes it. They are not strictly required; but are recommended to have. At the moment, Cycle Start doesn't do anything else.

Elpablito, 2019-11-30:
Hi
I have just a small hardware problem with the keypad. It is not clear why bat a button is not working, just one. Except this, the CNC and control are finished. Limits solved. I was not knowing you can use both. Probing and Part Zero I shall wait, no problem. 
An instruction generated by Freecad that GRBL do not like is “; M6 T9” or “; M6 T1”
Thank again for your support, if any news I shall report

bdurbrow, 2019-12-01:
Which key, specifically, isn't working?

bdurbrow, 2019-12-01:
Found & fixed the problem with comment handling; the bugfix is up on github now. The only change of consequence that I made was to the `SdFile_grbl.cpp` file; the changes to the `DriverUtilities.h` file shouldn't affect anything (but you can grab the latest version of that file as well if you like).

Elpablito, 2019-12-01:
Hi
Is the key “7” or “G”. I have changed the keypad thinking a trouble of the switch because 8 ad 9 was working and the same 1 and 4. The horizontal line and the vertical one are OK so I was thinking  “is the switch”, bat it is not. I have to find out, but it is strange. If you take away the front plate from the top of the box, it works. If you put the front plate again on the top, same time it works for a little then stop some time not. It may be a noise, but why just on 7? Could be a false contact, but why only on 7 and not on all keys of the same line?.  So the question is again why just on 7? I have to discover

Elpablito, 2019-12-01:
Hi
This is also interesting. I have a nasty problem with keyboard my suspicion was noise, but…. My setting is rotation of keyboard = 0. Swapping lines with columns i.e. inverting the connector of flat cable of keyboard, the keyboard is working properly, for sure with a different layout !!!!!!!!!!

Elpablito, 2019-12-02:
Hi
I am thinking that it is not only my box. I changed the length of cable connecting keyboard with Ramps, I twisted wires, I add a ferrite bed no way same result. Button 1 and A not recognized. But … pushing D, arrow on the top left, pushing 1 arrow is cancelled.  Reversing the connection keyboard is OK.

littledrumer, 2019-12-02:
I fixed my SD card issue if anyone else has the same problem.

To fix the error where inserting an SD card would reset the system I replaced the Smart Controller LCD.

To fix the issue I was having with running a file from an SD card by removing the % symbol at the beginning of the file as stated here
https://github.com/bdurbrow/grbl-Mega/issues/3#issuecomment-528932923
but I believe this issue has already been fixed in the latest update.

bdurbrow, 2019-12-03:
I think I'll add a keypad diagnostic mode that will be available as a compile-time-option... look for it tomorrow or the next day. It will send keypad presses to the serial port so you can see what the keypad driver is actually receiving from the keypad.

bdurbrow, 2019-12-03:
Keypad diagnostics are now up on github. Also, I updated the `Keypad.pdf` overlay to include functions for the RPN calculator.

littledrumer, 2019-12-03:
Great! Thanks for the update.

zvizdic, 2019-12-14:
It’s been a while since played with my cnc (no time).
I see some potentiometers for feed control and rpm on some setups.
Where do you connect them I like to add to my setup any help.
Thanks

bdurbrow, 2019-12-15:
On RAMPS, override pots connect to the headers that are used for thermistors on a 3D printer.

This wiki page has a pinout diagram:
https://github.com/bdurbrow/grbl-Mega/wiki/Wiring-Hookup-for-CNC-with-RAMPS-1.4

Also, you will need to enable them in config.h; look at around line 667.

Use 5k linear (not log or audio) pots; with one or the other fixed terminal of the pot going to ground, and the wiper terminal going to the signal input pin. If the value moves backwards from what is desired when the pot is turned, switch the ground lead to the other fixed terminal on the pot (it helps to test this with an alligator clip before soldering the wires to the pot).

zvizdic, 2019-12-15:
Thanks !
Now I need to learn how to enable feed in config.h

bdurbrow, 2019-12-15:
> enable feed

Not sure what you mean by that? You mean turning on the pot inputs in config.h?

If so, the lines in question will be commented out (in other words have a // in front of them so the compiler ignores them). Just remove the // to turn it on.

For example, this is what's in config.h starting on line 667
```
// If defined, this will enable the feed rate override potentiometer.
// The value specified is the ADC channel to be used. Input voltage range is 0 to 2.5 volts.
//#define UI_FEED_OVERRIDE_POTENTIOMETER		    13
```

Turning the option on would look like this:
```
// If defined, this will enable the feed rate override potentiometer.
// The value specified is the ADC channel to be used. Input voltage range is 0 to 2.5 volts.
#define UI_FEED_OVERRIDE_POTENTIOMETER		    13
```

Notice the lack of // before the #define.

Then, re-compile Grbl and download it to your Arduino.

zvizdic, 2019-12-15:
Thanks!
That is what I need.

zvizdic, 2019-12-16:
[image]

zvizdic, 2019-12-16:
My keyboard is upside-down how to fix that?

bdurbrow, 2019-12-16:
Either the keypad is plugged in backwards or the orientation in config.h is wrong.

Look on line 641 of config.h; there's an option there for setting the orientation of the keypad.

At one point, I had accidentally posted a version of config.h to github with the keypad orientation set to 180.... I'm just guessing but perhaps you had downloaded that version, built your machine with the odd config.h, and put the keypad on to match the behavior of it, so now it's "backwards"?

Well... that's what the keypad orientation setting is for... so you can put the keypad on any way you want.

bdurbrow, 2019-12-16:
Well, the latest version is always _supposed_ to be better.... unless I've messed something up.

What's the date on the previous version you had?

bdurbrow, 2019-12-16:
> just need some labeling to do!

Oh, and one other thing: the next update will have new keypad overlays - I'm adding some new functions to a couple of the keys. So, when you are doing the labeling, if you do the keypad, be sure it's not attached with permanent glue.

zvizdic, 2019-12-16:
I am not sure about version I’m lost witch is.
But it would be good to ad program start on pin D32 not only resume as is now.

zvizdic, 2019-12-16:
Maybe use reset button from reprap controller to reset program to beginning to resemble Fanuc controller.

bdurbrow, 2019-12-16:
> Maybe use reset button from reprap controller

That button is used by the UI layer as a general purpose "cancel" or "go back" button. I've got a wiki page that I'm writing that documents this (and other stuff). It'll go up on github with the next major update.

What I think I will add is a menu option to cancel the running of the current program... so the workflow would go something like this:

1: Press the Feed Hold switch to stop the machine from moving.
2: Press the Menu key
3: Select "Cancel Program"
4: Confirm the cancel (aborting a program in the middle of it's run could result in a ruined part, so it's best to double-check with the user).
5: Reset everything you need to (tool, clamps, perhaps put a new piece of stock in, etc)
6: Run the program again from the SD Card menu.

bdurbrow, 2019-12-16:
> But it would be good to ad program start on pin D32 not only resume as is now.

Yeah - I agree. That will be in the next major update.

tevins, 2020-02-10:
hi bdurbrow,
New to CNC,
I cann' t start the g-code programe I selected form sd card, I didn't found the button to do that job. Did you put program start on Pin D32? I push the cycle start button but it doesn't start the g-code programe, a guy said :"But it would be good to ad program start on pin D32 not only resume as is now." , from it  I deduced that a button called "resume" will do the work, but I check the grbl pinout on UNO resume and cycle start actually the same pin. Did I missed something? Please help, thank you!

BTW, you did a great job!  I am looking forward for your new updates!

Elpablito, 2020-02-14:
hi bdurbrow
I had to remake my CNC because you can not use plastic parts. A question; if I use the DRV8825 to drive a  stepper nema23 3A limiting the current to 2,5A , have I just a torque reduction ?

Saur0o0n, 2020-02-14:
For DRV8825 2,5A is probably too much (unless you have some specific version) - recommended current is 1,5A or 2,2A with heatsink AND forced air cooling.
So I wouldn't go above 2A with it.
Yes, you will loose torque - and everything depending on it like speed, acceleration etc.

Elpablito, 2020-02-14:
Hi Saur0o0n
Ok you are confirmed my doubts. Now the question is: there is an already tested solution to drive a 3Amp stepper using mega and ramps? If you give me a suggestion I will look for.

Elpablito, 2020-02-14:
Hi Saur0o0n
TB6560  may be a possible solution

Saur0o0n, 2020-02-15:
Hi,

So far I had 3 drivers (all Chinese production). Two (max 48V/5A)  I've burned without any particular reason - they stopped working properly (there where buzz), after a while one burned totally - those where based on TB6600. Chip inside seems to be real.
I had also "faked" TB6600 - this are based on smaller chip that cannot dissipate enough heat - that's why it's no go for 2A+. This one was working, but for small currents.
Now I have DM556 (this are based on ARM micro controller) - I can recommend them. They work nice, motors are running quite silently and on hold they can lower current (so less heat on motor).
One problem was that they are rated to 50V but they had 35V capacitor on main voltage - it blow to my face ;) almost immediately after connecting (my voltage is 48V) - but after replacing caps, they are fine. I bough them [here](https://www.aliexpress.com/item/4000339185928.html).

bdurbrow, 2020-03-05:
@tevins - Sorry about the delay; things got busy here and I haven't had much free time. As mentioned above:

> Select the .nc file, and then press the "Enter" key or push the Encoder Knob in. The file should then start running.

Other behavior (i.e, activating with the Cycle Start button) will be added in the next update.

Elpablito, 2020-09-09:
hi bdurbrow
I am back after long time becaure I had to remake the CNC and because all the problems of this year. I hope you are OK.
I am not able to discover how P+, P-, and Part are working. Could you give a suggestion  or say where I have to looK for ?

Elpablito, 2020-09-14:
Hi
I hope that covid19 is not the reason because no one has been on this forum since 5Mar.
There is some one who is able to explain to me how P+, P-, and Part are working ?

austrian1, 2021-04-27:
hi you did a great job there. still trying to figure out where should i connect the buttons.

austrian1, 2021-04-27:
Great job bdurbrow . the button reset although makes you go to menu and reset the alarm.  and one more when you run a program from sd card you see x,y, z changing, if you go to main menu afterwards i didn't find a way to go back to x,y,z without cancelling the program . maybe you could add an external button for coolant.