bdurbrow/grbl-Mega #22: Enhancement: add jog menu.

thawkins, 2019-10-27:
Add a jog menu to the lscd menues, with the ability to select an axis,  movemeht would track the ui encoder mivement (1mm/step x/y, 0.1mm/step z) until tbe encoder button is pressed to terminate jog mode.

bdurbrow, 2019-10-27:
Jog mode is currently triggered from the keypad; and the encoder button switches between jog increments (the increments available are configurable in `config.h`.

I can add an optional (via #define setting) set of menu commands to enter jog mode with a specific axis selected.

thawkins, 2019-10-28:
What about using long press on the encoder button to exit jog.

bdurbrow, 2019-10-28:
> What about using long press on the encoder button to exit jog.
The UI infrastructure isn't set up for that. Everything is done on key-down events, not key-up events.

Exiting jog is done with the second button on the LCD card (generally speaking, that button is used as "back" or "cancel").