bdurbrow/grbl-Mega #23: Enhancement: add overrides menu:

thawkins, 2019-10-27:
Add a menu to edit overrides,  have a submenu for spindle speed,  rapid feed and cut feed:  lcd ui encode allows override to be increased/decreased until encode button is pressed.

bdurbrow, 2019-10-27:
Well; I'll add it, but I do think that if it's at all possible having dedicated knobs (via 5k pots) is the better option - just because you can grab one in a hurry if you need to.

thawkins, 2019-10-27:
Im thinking about how usefull the system is without all the extra
hardware,  the reprap discount card is simple as it can be purchased and
there is no extra wiring required, the rest requires hardware and wiring
that may put off a lot of people.

You should load a current version of Marlin (3d printer firmware) onto a
2560/reprapdiscount display, and play with its menu system, its extensive
and very easy to use. I used it in my 3d printers for years, and it was
very simple.