bdurbrow/grbl-Mega #25: ENHANCEMENT, Add plasma THC compatibilty and functions

sinfocomp, 2019-12-15:
Can the functions described here : [http://linuxcnc.org/docs/devel/html/plasma/plasmac-user-guide.html#cha:plasmac-user-guide](url) be added ?, it would be a great enhancement 
thanks

bdurbrow, 2019-12-15:
Plasma cutter support is on the roadmap. I intend to support both up/down and analog arc voltage modes, and have a separate input for a nose-in-contact-with-workpiece sensor. I would suggest a microswitch with an adjustable probe tip instead of an "ohmic" sensor so that it can deal with workpieces that have a poor electrical impedance to ground (due to paint, rust, anodizing, etc).

I won't have it done until next year sometime, though... and the software module is going to be available before it's been tested on hardware here by me, because although I have a plasma cutter in the garage, I don't have a CNC table for it put together yet, nor do I have the air system to feed it setup yet (I've got the parts, but they're still in boxes). So, expect it to not work right on the first try.

sinfocomp, 2019-12-15:
i will be happy to test and help debug, i have a full size plasma and full size laser (both 1.5 by 3 meter)

erosenst, 2020-12-03:
Hello, any progress about THC?
Is it posible to loan functions from:
https://github.com/UnfinishedBusiness/XmotionFirmware/tree/master/src
main.c
And port to 2560?

I have plasma table and like to test functions. Currently run it grbl-mega-5x due X-axis cloning dosn't work in this sd-card version..
 Is it posible to use grbl-mega-5x instead grbl-mega at starting point from that sd-card support..?

erosenst, 2020-12-13:
Hello added LCD + SD-card +THC functions to mega-5X. Works atleast little but dosn't like G38.2 from sd-card, from usb same file execute normally. Not know yet why..

[grbl-Mega-5XLT.tbz.zip](https://github.com/bdurbrow/grbl-Mega/files/5684684/grbl-Mega-5XLT.tbz.zip)

erosenst, 2020-12-15:
There was pin conflict in cpu_map.h and added plasma voltages to display, Now it works atleast I can stream file from sd-card. New sourcepack:

[grbl-Mega-5XLT.tbz.zip](https://github.com/bdurbrow/grbl-Mega/files/5698221/grbl-Mega-5XLT.tbz.zip)

erosenst, 2020-12-19:
Added some compile options to disable useless menus/functions (for plasma cutter). Now compile 55.5kb diet version so sd-card + lcd support is only 19kb..
 If you use/test it can you comment here?
known bugs is g-code error number is not shown correctly to LCD and it sends probe (G38.2) data to serial when run sd-card..

[grbl-Mega-5XLT_v0.1.tbz.zip](https://github.com/bdurbrow/grbl-Mega/files/5720206/grbl-Mega-5XLT_v0.1.tbz.zip)

sinfocomp, 2021-01-25:
happy to test, will get back soon

sinfocomp, 2021-01-25:
> Added some compile options to disable useless menus/functions (for plasma cutter). Now compile 55.5kb diet version so sd-card + lcd support is only 19kb..
> If you use/test it can you comment here? [...]

how to conect ? and where to setup THC settings?
Regards

erosenst, 2021-01-28:
how to conect ? and where to setup THC settings?
Regards
--
connect with usb and give settings like normal grbl..

THC voltage is analog pin 15 (if rember correct) In my board it is bed temp. Pullup resistor needs to unsolder to use with THC cap can be there. Spindle speed sets THC voltage, range some 700-1023 that converts to A15 pin ~0.6 - 1.1v range. I use 10Mohm high volt resistor coupled to 150kom resistor. so 0.015 multiplier.

Helps this any? Got any numbers to LCD?

sirias52, 2021-05-16:
Hi erosenst. Could send more detail about your proyect. for examle, any conextion diagram.

erosenst, 2021-05-19:
Hello, you can found connection pins in cpu_map.h. Use as normal ramps board. Base is grbl 5x + bdurbrow lcd+sd-card code. I use it in two machines currently. One is double x-axle machine and another is real 5x machine.
 Bigtreetech sd-cloud wlan adaptor works fine with nautilus and I can write files over wlan to machine then run them with lcd knob..

Have you compiled it and loaded to ramps board?

sirias52, 2021-05-26:
I get it. Do you have any diagram to create voltage divider whithout damage analog input to arduino?

erosenst, 2021-05-27:
I have two coils (~10r around big EMI ferrite) connected to cutter, then 10Mohm high volt resistor + 150ohm resistor. Paraller to 150ohm resistor is paralled three 3.3v 1w zener (one is enough but I put three) and some cap. There I connect cox to arduino A15 (bed temp) and there is little coil near A15 to supress emi. This all seems to be enough and 2650 1.1v see some 0.65-0.95v volts in my setup when arc is on and HF is not grounded due two coils so it ignites fine..  Arduino is grounded through three diodes (0.6vx3) + one more difrent direction + eletrolytic cap to float little arduino ground potential..

miamijerry, 2022-07-19:
> how to conect ? and where to setup THC settings? Regards

I have tried to install this version
[grbl-Mega-5XLT_v0.1.tbz

But I get a number of Compile issues;
Line 17 in DriverUtilities.h needs to be changed to;
//typedef uint8_t boolean;
and
All boolean need to be changed to bool
Then it compiles and installs

But the 4 X 4 Matrix keypad is not working??
Any advice??

miamijerry, 2022-07-20:
OK, I edited DriverUtilities.h lines 16 and 17 to;
typedef bool boolean;
//typedef uint8_t boolean;

Both grbl-Mega-5XLT_v0.1.tbz and grbl-Mega-5XLT.tbz install fine now.
But grbl-Mega-5XLT_v0.1.tbz does not have a working keypad...