bdurbrow/grbl-Mega #27: LCD12864

agap19, 2020-06-03:
How to enable the 12864 screen in the firmware. Help is very necessary, you can e-mail agap19@mail.ru thanks

bdurbrow, 2020-06-20:
Sorry about the delay; things have been crazy here.

As mentioned in the main ReadMe, this does not support graphic LCDs (i.e, 12864 or other similar devices).

> A RepRap Discount Smart Controller LCD interface w/ Smart Adapter (the one with the 20x4 character LCD, rotary encoder w/push button, extra push button and SD card slot; NOT the ones with a graphic LCD, this firmware doesn't - and won't - support those).

You will need to acquire compatible hardware in order to make it work.

ToxicDeathX, 2021-03-01:
> Sorry about the delay; things have been crazy here.
>
> As mentioned in the main ReadMe, this does not support graphic LCDs (i.e, 12864 or other similar devices).
>
>
> You will need to acquire compatible [...]

Sorry to necro this issue would like to know what would be the technical specifications and or what is causing the LCD 12864 to not be supported, i know you said that you don't want to support it but what issues did you run into if any with trying to get the LCD to work?