bdurbrow/grbl-Mega #31: Jog mode bug

how-eee, 2021-03-16:
for example
activeIncrementValue = 1.0000
when I turn encoder A or encoder B (if present) slowly? position incremented by 1
If I turn encoder rapidly back an forth, position incremented accidently by decimal number (for ex. 0.125)
If USE_UI_ENCODER_B_ISR enabled or disabled nothing changed. same issue.
I tried many different encoders, mpg wheels. Its software issue.
I tried to fix it but cant understand UISupport.cpp . My programming abilities too low for it =(
please help...

how-eee, 2021-03-20:
EXEC_MOTION_CANCEL flag set when changing direction while axis moving

how-eee, 2021-03-22:
also still cant figure out why А feedrate didnt change actual speed in jog mode
maybe im doing something wrong?
could you tell me how do you use it?

how-eee, 2021-03-23:
I have added feedrate change during jog. As it used during PowerFeed
config.h
#define UI_JOGGING_BASE_STRING_METRIC   		"$J=G91F"
#define UI_JOGGING_BASE_STRING_INCH     		"$J=G91F"

UISupport.cpp
if(!UILineBufferIsAvailable()) return;

    const char *source = (gc_state.modal.units)?PSTR(UI_JOGGING_BASE_STRING_INCH):PSTR(UI_JOGGING_BASE_STRING_METRIC);
    char *cursor = UILineBuffer;
    char c;
    while ((c=pgm_read_byte(source++))) *cursor++ = c;
    **cursor = sPrintFloat(cursor, 10, jogFeed, 1); //set feed** 
    *cursor++ = activeAxis;
    
    cursor = sPrintFloat(cursor, 10, activeIncrementValue * count, 4);

    *cursor = 0;

Also fix a bug of feedrate. It didnt change value if editBuffer<4
 
void JogUIPage::enterKeyPressed(void)
	{
		if(!editingJogFeed) return;
		
      if(editBufferState == 0)
      {
        jogFeed = 0.0;
      }
      else
      { 
        **while(editBufferState<4) {editBuffer[editBufferState++] = ' ';}** 
    		jogFeed = stringToFloat(editBuffer);
       }

rikstablades, 2022-03-02:
> EXEC_MOTION_CANCEL flag set when changing direction while axis moving

________________________________________________________

Where did you locate this in the script? Which Line(s) please?

Kind Regards

how-eee, 2022-03-06:
UIsupport.cpp 
line 1863
line 1889
and few times later i this file

But I forgot already if the problem still there. it was year ago/ I revised all code and post branch there https://github.com/how-eee/grbl_lathe_UI_threading 
My lathe still working every day with no issues (minor issues dont count)

rikstablades, 2022-03-06:
> UIsupport.cpp
>
> line 1863
>
> line 1889
>
> and few times later i this file
>
>
>
>
>
> But I forgot already if the problem still there. it was year ago/ I revised all code and post branch there [...]

*************************

Many thanks for your reply...

Kind regards

rikstablades, 2022-03-08:
> *************************
>
> Many thanks for your reply...
>
> Kind regards

Hello 

I've looked at this UISupport.ccp and your UISupport.ccp on your branch and can't see any difference between the two.

Did you rectify this issue?

I've noted that the Jog Wheel has a tendency to change direction when spun quickly which cancels the movement very briefly mid Jog and gives rise to the incomplete travel/movement, hence odd values.

Also, the wheel doesn't respond to the first 'dent' or click when direction is changed, you get like a dead spot. This only happens when direction has been changed though.

Is this an issue in the Protocol script? 

Thoughts welcomed...

Kind regards

how-eee, 2022-03-11:
I assume this is correct behavior.
When you jogging your mashine in one direction than rotate encoder another direction it  stopped immidiatly with EXEC_MOTION_CANCEL flag. And it doesnt lost its position.
If yoг wait until jog motion finished and then rotate wheel to another direction, first click change direction,second click move axis.
When you rotate encoder fast it starts to jitter, and you get wrong series of impulses of your encoder.
I made 3dprinted own MPG with two latched hall sensors and magnets. It works perfect at any rotation speed. I dont have issues with it anymore.

how-eee, 2022-03-11:
you can see it here https://youtu.be/8M4gzLkSXls

rikstablades, 2022-03-12:
> you can see it here https://youtu.be/8M4gzLkSXls

Hello How-eee 

Thanks for the link and reply.

Great job on the DIY Encoder. Works really well it seems.

Following your comments, I've added 10nF caps to the lines on my MPG wheel and it has cleaned up the movement nicely!

Thanks for sharing...

Kind regards

lgojniczek, 2022-05-07:
> Hello How-eee
>
> Thanks for the link and reply.
>
> Great job on the DIY Encoder. Works really well it seems.
>
> Following your comments, I've added 10nF caps to the lines on my MPG wheel and it has [...]

How did you connect those caps?

rikstablades, 2022-05-08:
> How did you connect those caps?

Between both lines and GND

Google 'Rotary Encoder caps'

Kind regards