bdurbrow/grbl-Mega #34: Is the dev on this dead?

d4mer, 2021-04-19:
Hi all,

Is the dev on this dead? Is this useable?

how-eee, 2021-04-21:
Hi there were a few bugs and errors.
I have it fixed.
Also I added threading G33 support.
There are still some minor issues, but everything is works perfect on my lathe.
You may check my fork here https://github.com/how-eee/grbl_lathe_UI_threading
Videos are also available there.
Feel free to ask questions on my page. Bdurbrow does not respond here.

bdurbrow, 2022-08-04:
I'm alive - but I haven't had any time to work on personal projects like this recently... and I'm pretty sure it's not gonna ease up for a few more months at least.

bdurbrow, 2024-01-25:
Still not dead - but although it's going to take a while to get back in the swing of things, I **have** unburied some of the machines and started on rebuilds. My water heater exploded - well, it didn't _explode_ in the same way that the one in that episode of MythBusters did, but it did vent thru the overload valve filling the garage full of steam. Again. This had the predictable result of rusting most of the cast iron surfaces in there. Oh well... I always wanted to learn hand-scraping anyways, I guess?