bdurbrow/grbl-Mega #36: nothing works?

Tonybeta01, 2021-10-30:
Hi have just set up a mega with a 2 differant ramps 1.6 boards and the 20x4 reprap controller.
I have succesfully loaded the firmware on . all the parts have been used and I know there working
no mpg encoder connected or keypad.
The lcd backlight comes on and the steppers lock.
Any Ideas
regards
Tony

rikstablades, 2021-10-30:
Steppers locking is normal, type '$SLP' into serial monitor to see them unlock.

LCD not working means pinout definitions for your setup are not correct. Change them.

Tonybeta01, 2021-10-30:
I haven't modified any code , I have noticed occasionally when I plug it in it comes up with garbage on the screen, a missmash of letters and figures
Regards 
Tony

⁣Get BlueMail for Android ​

Tonybeta01, 2021-10-30:
ok I wound it up to 15 vols and it works now?? Ican bring it down to 12v and its still OK, Ive put a keypad on it but its a bit buggy. for instance how do you jog an axis?
regards Tony