bdurbrow/grbl-Mega #39: mcodes to switch pins on and off

Tonybeta01, 2021-11-10:
Hi, id like to be able to associate 2 mcode numbers to switch a pin high and low, to control relays/solenoids ect, is this possible? ive done it in mach3.
Regards
  Tony

Ams-Laser, 2021-11-20:
im too looking for this option

Tonybeta01, 2022-02-07:
Can I use M42 to change a pin condition?
also can I read a pin?

Tonybeta01, 2022-02-14:
as per this:
https://marlinfw.org/docs/gcode/M042.html

is there a quick and easy way to find out which pins are free?

Saur0o0n, 2022-02-14:
[This](https://github.com/Saur0o0n/GRBL_Mega-pinout) should be easy enough :)

Tonybeta01, 2022-02-14:
so M42 is not supported , thats a shame

Ams-Laser, 2022-04-15:
Heelo, how hard would it be to pull this update ?
https://github.com/fra589/grbl-Mega-5X/commit/ec0471874f7909a643b2fc30bb2e414011d926f4