bdurbrow/grbl-Mega #40: CNC plasma cutter

Tonybeta01, 2022-03-06:
Hi after trials and tribulations I have managed to set this up with everything working, and I really like its potential, shame about not beig able to use M62 though.
   To my question... I have started building a portible CNC plasma cutter, I was initially going to use it with mach 3, but wouldnt mind having a go with this controller.
   Has anyone done this yet? if so is there a chance you could give a few pointers? it would be great if I could have a THC with it, even if using using a second linked  arduino?
   Any thoughts?
  Regards
  Tony

how-eee, 2022-03-06:
Give it a try! Go ahead.

Tonybeta01, 2022-03-06:
I was hoping for advice?

Ams-Laser, 2022-03-25:
> I was hoping for advice?

Hello, if you use shielded cabling and non HF plasma works great, also try using proma-sd THC